SimpleMobileTools/Simple-Camera #23: Remember last used camera/add option to configure default camera

Lesik, 2016-10-10:
Hey,

first of all, great software. I like using it and the fact that it's libre software makes the icing on the cake.

It would be great if the app would remember the camera that was last used, whether it was the front or the back camera. Or, if you oppose this change, a config in the settings to set the default camera would do it, too.

My smartphone's back camera broke and all camera apps I tried before crash because they can't access the camera, before I can even switch to the front camera (which works fine). But Simple Camera doesn't crash. However, since the back camera is essentially useless for me, I'd love if Simple Camera could automatically start with the right camera setting.

Thank you in advance.

tibbi, 2016-10-10:
will be added in 1.22, which has just been published

closed 2016-10-10 by tibbi

Lesik, 2016-10-12:
I just received the update through F-Droid. Thanks for that useful feature, @tibbi!