SimpleMobileTools/Simple-Camera #44: App crashes when switching to video-mode

x-jokay, 2017-02-20:
Using Gallery version 2.3.9 on two Sony Stock ROM's (Android 6.0.1 and 7.0.1) when switching from "photo-mode" to "video-mode" the app crashes.

tibbi, 2017-02-20:
How is the gallery related? I dont get it :)

x-jokay, 2017-02-20:
:D my fault, I mean the Camera app, sorry ;)

x-jokay, 2017-02-20:
My fault I meant the camera app, sorry ;)

tibbi, 2017-02-20:
Yea the Camera has many issues as theres a lot of things done differently per manufacturer and per OS version. That causes a lot of issues that I just cannot reproduce on any device, so I cannot even fix yet. Hope it will be improving over time, one bug at a time.

x-jokay, 2017-02-21:
Ok, I understand.

Please let me know if I can help to resolve the problem or test a version creating some debug output (on a not rooted device) which may help.

tibbi, 2017-03-22:
Hey, Ive just released a new version 2.0.0, please try if its better. Thanks

x-jokay, 2017-03-24:
Thank you, I will test it as soon as version 2.0.0 is available on fdroid.org (https://f-droid.org/wiki/page/com.simplemobiletools.camera).

x-jokay, 2017-03-27:
I can confirm using the version 2.0.1 of the camera app on a Sony Stock ROM (Android 7.0) everything works.

Thank you very much, issue can be closed.

closed 2017-03-27 by x-jokay