SimpleMobileTools/Simple-Camera #51: "Unfortunately, Camera has stopped" when taking picture

mugpahug, 2017-05-02:
On Android 4.1 I get the error "Unfortunately, Camera has stopped" every time i take a picture. No image is stored.

tibbi, 2017-05-04:
Hello, if you go in Settings, what Save directory do you have selected? DCIM? Can you try changing it? Thanks

mugpahug, 2017-05-05:
Hi,

Yes save directory is internal/DCIM,

Changing it to any folder on internal/ still causes the app to crash 
when I take a picture.

When I change it to a directory under /root/ (such as cache or 
programs)  the app doesn't crash anymore when I take a picture. However, 
the image doesn't get stored in these folders (I guess it has to do with 
permissions or so?)

I hope this helps, if you need more info please let me know.

Cheers

tibbi, 2017-11-05:
this is an old report, if it still isnt fixed, it should be fixed in the next version.

closed 2017-11-05 by tibbi