SimpleMobileTools/Simple-Camera #57: Don't persist 'Always' flash

smichel17, 2017-06-10:
Occasionally I want to force the flash on. When I come back to take a picture a day later, I seldom want the flash on and my first action is always to turn it off. Can we not persist this setting? Details:

- When I turn the flash off (not auto), I would not appreciate it reverting to auto. 
- When I force the flash on, if I quickly exit and return to the app, I would not appreciate having to turn the flash on again.

Therefore, I suggest the implementation of: when the app is opened (ie, in onCreate), if the persisted value is `always on`, change it to automatic.

tibbi, 2017-06-14:
not sure whats the issue there. Whatever is the flash state and I quit the app, the flash state should be the same as at quitting. Thats what the user would expect, I dont want to do any shady things in the background.

trubitsyn, 2017-06-14:
As @smichel17 said, sometimes it's confusing to have flash suddenly turned on. Could you think on an option for that (turned off by default)?

tibbi, 2017-06-14:
ye I (you) could add a toggle in settings to "Turn flash off at startup", which would be disabled by default. If enabled, in onCreate it would be turned off in all cases. That should be good enough, I dont want to overcomplicate it.

trubitsyn, 2017-06-14:
@tibbi, please look at #58. I just realized I used a different option name than yours. Should I rename "Reset enabled flash to auto on exit" to "Turn off flash at startup" ?

Thanks

trubitsyn, 2017-06-14:
@tibbi I followed your instructions and now everything works as you requested.

commit c8e4f411 (2017-06-14): Merge pull request #58 from trubitsyn/master

Fix #57
commit ebd53d94 (2017-06-14): Fix #57
closed 2017-06-14 by tibbi