SimpleMobileTools/Simple-Camera #59: Use Camera2 API on Android 5 and greater due to Camera deprecation

trubitsyn, 2017-06-14:
Reference: https://developer.android.com/reference/android/hardware/camera2/package-summary.html

tibbi, 2017-06-14:
i wont change it anytime soon as I dont want to maintain 2 codebases, but I can leave it here for the future

cryptomilk, 2018-02-18:
I think most phones offer a HAL3 now. So it would be nice to have an app supporting API2 will allows you to do much more settings, like ISO, AWB or EV control.

Also it would allow to load QCOM extensions/filters.

See:

https://source.codeaurora.org/quic/la/platform/packages/apps/SnapdragonCamera/tree/src/com/android/camera/imageprocessor/filter?h=camera.lnx.3.0.r2-rel

tibbi, 2018-06-11:
The app will be using Camera2 from version 4.1

closed 2018-06-11 by tibbi