SimpleMobileTools/Simple-Camera #66: small Bug in Front camera zoom and auto focus

deepankumarp, 2017-08-04:
I tried your code, some bugs are there.
Front camera open and zoom it after touch the screen,app was stoped.

The errors are below,
com.simplemobiletools.camera, PID: 21850
                                                   java.lang.RuntimeException: setParameters failed
                                                       at android.hardware.Camera.native_setParameters(Native Method)
                                                       at android.hardware.Camera.setParameters(Camera.java:1956)
                                                       at com.simplemobiletools.camera.Preview.focusArea(Preview.kt:350)
                                                       at com.simplemobiletools.camera.Preview.focusArea$default(Preview.kt:328)
                                                       at com.simplemobiletools.camera.Preview$2.onClick(Preview.kt:98)

deepankumarp, 2017-08-07:
Hi tibbi, what about above bug issue? I want your Gmail ID for future contact.

tibbi, 2017-08-07:
Hey, I will take a look at it once Ill be working on the camera, its not a priority right now.

deepankumarp, 2017-08-07:
I need your mail id

tibbi, 2017-08-07:
its hello@simplemobiletools.com

tibbi, 2017-09-21:
retry with 2.3.5 please which Im releasing in a moment

tibbi, 2017-10-21:
I will be releasing a new version within 1-2 days, instead of crash it will maybe just show some error. Let me know if its better. Thanks

tibbi, 2017-11-05:
it should be fixed in the next version released today

closed 2017-11-05 by tibbi