SimpleMobileTools/Simple-Camera #69: Geolocation

Michele64, 2017-08-18:
Would be nice to have it as an option

tibbi, 2017-08-18:
that requires location permission which many people wouldnt like, but I might add it as an option in the future

Michele64, 2017-08-18:
Thanks, maybe switched off by default so that android's request kicks in only when turning it on?

tibbi, 2017-08-18:
thats how it would work on Android 6+, but it would always be on on Android 4 and 5

Michele64, 2017-08-18:
You're right, and thanks for considering it

Yang-zLab, 2018-11-30:
I love simple camera, I use it as my work camera. It can save the photos to another diraction, so I can seperate photos for work from my personal daliy life photos. It can choose a lower resolution so the space of the phone can be saved. The only thing I am looking forward to see is the GPS metadata. If the location could be recorded, this app is perfect for work usage.

alecpetrosky, 2020-10-24:
Yes, completely agreed that this funcion is very handy for sorting out photos years later.

inson1, 2023-09-26:
I dont like giving more permissions than is necessary

Michele64, 2023-09-26:
but it would be the user's choice. And without that functionality many users will abandon the app, possibly frustrated because they realize it much later that their pictures lack an essential information....

inson1, 2023-09-26:
I get it. But there should be more opinions.

And I never used it and I dont like giving to message, photo more informations than needed.

I love options, but its also unsafe even as options.

Cloud drive storages and other software LOVES metadata.

And there are many things in backlog already.

What I am trying to say is that I would like to see more other things before. And If there is everything done then maybe then.

But thats my opinion. 1 vs 5

Saijin-Naib, 2023-09-26:
Yep, I have stopped using it because of this.

tibbi, 2023-09-27:
geolocation has actually been added already, so closing it. It is disabled by default and the app doesnt have location permission without toggling the setting and granting the Location permission.

closed 2023-09-27 by tibbi