SimpleMobileTools/Simple-Camera #73: Add QRcode reader

sheychen290, 2017-09-25:
It's so usefull and logic (IMO) to place a qrcode button in a camera app.
Thank

F3000, 2018-02-11:
it would be greate when it is like the ios11 feature ... open Camera -> hold camera over qr-code -> app display a little pop-up window with the informations decrypted in the code ->user can click to open it (e.g. a link)

Klimmbimmel, 2018-02-20:
Nah, not needed, other things are more important. Just use Firefox with its integrated qr-button

thomas-mc-work, 2020-11-09:
@Klimmbimmel There are other use cases like reading WI-FI credentials that Firefox cannot satisfy.

Pyromane, 2024-01-09:
That would be a great benefit!

inson1, 2024-01-09:
@Pyromane SMT apps were sold to ad company. https://github.com/SimpleMobileTools/General-Discussion/issues/241
Fork is here https://github.com/FossifyOrg

Pyromane, 2024-01-09:
> @Pyromane SMT apps were sold to ad company. [SimpleMobileTools/General-Discussion#241](https://github.com/SimpleMobileTools/General-Discussion/issues/241) Fork is here https://github.com/FossifyOrg

Thank you for the important information!