SimpleMobileTools/Simple-Camera #78: Added ja (Japanese) translation

RACER-JP, 2017-10-12:
I have translated your application into `ja` (*Japanese*). Here is the xml:
```xml
<resources>
    <string name="app_name">単機能ｶﾒﾗ</string>
    <string name="app_launcher_name">ｶﾒﾗ</string>
    <string name="camera_unavailable">ｶﾒﾗは利用できません</string>
    <string name="camera_open_error">ｶﾒﾗへの接続時に不具合が発生しました</string>
    <string name="video_creating_error">動画ﾌｧｲﾙの作成時に不具合が発生しました</string>
    <string name="video_mode_error">動画撮影への切り替えに失敗しました</string>
    <string name="save_error_internal_storage">不具合により､保存ﾌｫﾙﾀﾞｰは内部記録領域に変更されました｡</string>
    <string name="camera_switch_error">ｶﾒﾗの切り替えに失敗しました</string>
    <string name="no_permissions">ｶﾒﾗや記録領域の権限が許可されなければ､このｱﾌﾟﾘｹｰｼｮﾝは機能しません｡</string>
    <string name="no_audio_permissions">動画を記録する為､録音の権限が必要です｡</string>
    <string name="no_gallery_app_available">利用可能な画像閲覧ｱﾌﾟﾘｹｰｼｮﾝがありません</string>
    <string name="photo_not_saved">この画像は保存されていません</string>

    <!-- other aspect ratio -->
    <string name="other">その他</string>

    <!-- Settings -->
    <string name="save_photos">画像と動画の保存先</string>
    <string name="show_preview">撮影後に画像の概要を表示</string>
    <string name="shutter_sound">撮影効果音</string>
    <string name="back_camera">背面ｶﾒﾗ解像度</string>
    <string name="front_camera">前面ｶﾒﾗ解像度</string>
    <string name="photo">画像</string>
    <string name="video">動画</string>
    <string name="focus_before_capture">撮影前に再度焦点を合わせる</string>
    <string name="volume_buttons_as_shutter">音量ﾎﾞﾀﾝで撮影</string>

    <!-- Strings displayed only on Google Playstore. Optional, but good to have -->
    <!-- Short description has to have less than 80 chars -->
    <string name="app_short_description">閃光､拡大機能付きのｶﾒﾗです｡広告はありません｡</string>
    <string name="app_long_description">このｱﾌﾟﾘｹｰｼｮﾝは､静止画と動画の両方が撮影でき､前後のｶﾒﾗを切り替えたり､保存場所を変更したり､解像度を制限することができます｡閃光を利用したり､2本の指で拡大したりする事もできます｡動画の撮影中は､閃光を懐中電灯として使用できます｡端末の撮影ﾎﾞﾀﾝを押してこのｱﾌﾟﾘｹｰｼｮﾝを起動したい場合は､標準ｱﾌﾟﾘｹｰｼｮﾝを無効にする必要があります｡広告や不要な権限は含まれていません｡ｵｰﾌﾟﾝｿｰｽで､提供しています｡このｱﾌﾟﾘｹｰｼｮﾝは､大きな一連のアプリの一つです｡他のｱﾌﾟﾘｹｰｼｮﾝはhttp://www.simplemobiletools.comで見つけることができます｡</string>

    <!--
        Haven't found some strings? There's more at
        https://github.com/SimpleMobileTools/Simple-Commons/tree/master/commons/src/main/res
    -->
</resources>

```

---

*Translation made with [Stringlate](https://github.com/LonamiWebs/Stringlate).*

tibbi, 2017-10-12:
Hey, there is already a Japanese translation at https://github.com/SimpleMobileTools/Simple-Camera/blob/master/app/src/main/res/values-ja/strings.xml , can you please check it out, press the Pencil button at the top right corner of the file, edit whats needed, then submit your changes with some commit message? All strings are different in your translation, any idea why is the current translation so different? Is it a different dialect or something? Thanks

tibbi, 2017-11-05:
Ive updated some strings which werent updated, havent updated existing translations. Not sure if they are wrong in any way, or which is better.

commit 6247d67f (2017-11-05): fix #78, updating some Japanese strings
closed 2017-11-05 by tibbi