SimpleMobileTools/Simple-Camera #79: File upload shortcut - picturemode

nektor211, 2017-10-21:
Hello,

Could you please add an option to go directly to the picture camera (not video) when using file upload (like in html file input field)? Now it's posible to open camera for taking videos and change to picture mode, but it would be nice to be able to do it directly.
[image]

tibbi, 2017-11-05:
it happens only at VIDEO_CAPTURE intents, which is supposed to return a video. That's why you see the Camcorder app at a selection too. I dont want to change that behaviour. The Camera is set to photo mode only if the website requests a photo, it works fine.

closed 2017-11-05 by tibbi

nektor211, 2017-11-05:
Thank you for the explanation, I've fixed the web form (by adding `capture="camera" accept="image/*"` in case anyone else comes accross this issue) and it works fine.