SimpleMobileTools/Simple-Camera #95: Can't Save to External SD Card

RandolphNimmer, 2017-11-07:
seeing "open failed: EACCESS (permission denied)" toast when trying to take pictures after setting save directory to SD card location. I think you may have not included the WRITE EXTERNAL STORAGE permission in your app? 

I'm on an S4 with android 4.4.2

tibbi, 2017-11-07:
it is included, SD cards just arent handled properly on Android 4.4. Google made it really difficult to handle them, I havent found any way of solving it yet as I dont have a device for reproducing it.

F3000, 2018-02-11:
@tibbi is that possible on Android 7/8 ?

tibbi, 2018-02-11:
well it should work there, are you getting the error?

F3000, 2018-02-11:
Oh sooty . I have set up the wrong settings

tibbi, 2018-02-26:
it should be fixed in the next version for Android 5+, I cannot fix it for lower ones.

closed 2018-02-26 by tibbi