SimpleMobileTools/Simple-Camera #109: Make sure the photo is captured in the selected resolution

tibbi, 2018-01-15:


Klimmbimmel, 2018-02-20:
What's the problem?

tibbi, 2018-02-20:
according to some reports sometimes the photo is saved in a different resolution than the chosen one

Klimmbimmel, 2018-02-25:
Is it reproducable? Anything specific?

tibbi, 2018-02-25:
I cannot reproduce it. Whatever resolution the user selected, it was captured in 8 mpx or so I think

badener95, 2018-03-29:
I have this issue too. My camera is anyway a bit wonky because I run a custom rom, but this is the only camera app I came across with this bug. Log: https://drive.google.com/file/d/138snc4jfj_RqkM8eBCm4IVOYMzPM6you/view?usp=drivesdk

tibbi, 2018-03-29:
I dont think theres anything useful in those logs, at least they are too long :/ Anyway, so you see a bigger list of resolutions, but whatever you pick, the image is always saved in one? Does selecting a resolution at least change affect the preview? I mean if it is lower quality, or different aspect ratio, as appropriate. Does changing video resolutions work? Thanks

badener95, 2018-03-30:
I can choose between 11 different resolutions. Changing between 4:3 and 16:9 resolutions also changes the preview accordingly. But no matter which resolution I pick the result is always a 1920x1080 picture. I am sorry that the log was no help. I don't mind trying to fetch a better one. What would you advice to do in order to have a more helpful log?

tibbi, 2018-03-30:
I dont think a log would help, but if you are willing to test some debug versions you can shoot me an email at hello@simplemobiletools.com where I could send you an .apk which you then install (it is a lot easier than sending via github). I could add some debug messages there and you would tell what is written on them. Thanks

tibbi, 2018-06-11:
The Camera has been rewritten from scratch under the hood in version 4.1.0, please update whenever it becomes available and create a new ticket if you are still having issues.

closed 2018-06-11 by tibbi