SimpleMobileTools/Simple-Camera #116: Can't record if the phone is muted

Klimmbimmel, 2018-02-07:
I wanted to block the sound, when video recording is started. There was no setting for that, so I muted the phone. In the case of full silence (don't disturb mode) the recording didn't start and the app froze. In case of only vibrate, pressing the record button increased the volume and the annoying sound, which I don't want to record, was played again.
(Seems like the native camera app has the same issue. I'm on lineageOS 14.1., without any Google (no playstore, no play services)

Embridioum, 2018-03-30:
Same problem with stock Sony Oreo. Video camera does not have Do Not Disturb access so recording can not start.

tibbi, 2018-06-11:
The Camera has been rewritten from scratch under the hood in version 4.1.0, please update whenever it becomes available and create a new ticket if you are still having issues.

closed 2018-06-11 by tibbi