SimpleMobileTools/Simple-Camera #120: Slow Preview on Nexus 4

frankFic, 2018-02-22:
Hi there,

I use simple camera on my Nexus 4 and found out the preview is very slow. I tested the frame rate is only around 10.
I also found Stack Overflow discussions on similar issue of low frame rate on this device:
[Stack Overflow Post 1](https://stackoverflow.com/questions/20252352/image-preview-is-very-slow-on-nexus4)
[Stack Overflow Post 2](https://stackoverflow.com/questions/14131900/extreme-camera-lag-on-nexus-4)

The workaround is to set recording hint to true before entering the preview.
If you would like to, I may submit a pull request.

tibbi, 2018-02-22:
well you can try it on your device. If it fixes the issue just submit it. I can then test it on my devices and if it doesnt break anything, guess it will be save to add. Thanks

frankFic, 2018-02-22:
@tibbi Thank you for your reply! I am not 100% sure where to insert the workaround. According to my current understanding, I inserted the workaround in `Preview.initCameraPreview` before `mCamera!!.parameters = mParameters`. Is that correct? Should I also need to call `setRecordingHint` at other places?
It seems this can fix the issue on my phone and I may submit a PR first for your review.
Thanks!

tibbi, 2018-02-22:
ye looks ok, Ill just tweak the code a bit. Check if it works even after toggling video, flashlight or so. I tried it on my device but it lowered the quality, so I will leave that Model check there. The Camera code is pretty messed up indeed, Ill rework it someday. Thanks

commit 1c4543f4 (2018-02-22): Fix #120
commit 513c3896 (2018-02-22): Merge pull request #121 from frankFic/master

Fix #120 (slow preview on Nexus 4)
closed 2018-02-22 by tibbi