SimpleMobileTools/Simple-Camera #145: User is not informed when taking a photo

F3000, 2018-05-15:
Please add a little white flash after taking a photo .
I have my phone muted and don’t see any action when I take a photo

tibbi, 2018-05-16:
well you can see the bottom icons disappear for a moment

F3000, 2018-05-16:
I haven’t recognized it yet. Maybe you can implement this is future releases

MisterY, 2019-02-07:
I would second this request. The sound is too loud (not configurable) for indoor shooting so I keep it off. But then there is no obvious sign that a photo has been taken.
Will try the recommendation above but it would be great if there was some explicit visual sign on the screen.
It could be a flash, image border, animation, anything really.

MisterY, 2019-02-07:
The icons disappearing is not bad but also not obvious to new users. People tend to focus on what's on the camera viewer, not on the buttons under their fingers. That is not intuitive at all, unfortunately.

commit 7e4985d6 (2019-09-03): fix #145, blink in a black screen at photo capturing
closed 2019-09-03 by tibbi