SimpleMobileTools/Simple-Camera #160: Camera button does not work

pager72, 2018-05-27:
My phone, like many others have a dedicated button to launch camera and take photos. Currently the app does not recognise this button and pressing it does not take photos.

tibbi, 2018-05-27:
If you want to launch this app at pressing the hardware camera button, you might have to disable the built in Camera app in Settings -> Apps -> Camera -> Disable. If it doesnt work, does it work at any other camera than the stock one?

pager72, 2018-05-27:
The trouble is the app doesn't take photos using the camera button. Launching the app with a camera button is not the issue.

tibbi, 2018-06-11:
The Camera has been rewritten from scratch under the hood in version 4.1.0, please create a new ticket if you are still having issues.

closed 2018-06-11 by tibbi