SimpleMobileTools/Simple-Camera #170: Video start button closes app

mrvanes, 2018-06-15:
When pressing the start videorecording button, Simple-Camera closes by sliding down out of view, no FC messages from OS.
Phone: Alcatel Idol 2 Mini S (6036Y)
Android: 6.0.1
Kernel: 3.10.49

tibbi, 2018-06-15:
what app version do you have?

mrvanes, 2018-06-15:
4.1.1 12 june 2018.

tibbi, 2018-06-18:
not sure what is the issue, but instead of crashing a message should appear in version 4.1.2 released soon. Let me know if you see it and what is written on it :) Thanks

Klimmbimmel, 2018-06-23:
Might changing the resolution help?

mrvanes, 2018-06-25:
Already tried that. No improvement.

GarciaPL, 2018-07-28:
I would like to take a note that in my case video start button also closes app when videos are set to be saved on SD card. If this setting is changed to save videos on internal memory, everything works as expected.

mirgleich, 2018-08-08:
I got the same Issue (probably a second one too).
I had the same behaviour as @mrvanes when saving to SD Card
When trying @GarciaPL s solution I'm getting a Java.lang.IllegalArgumentException: Surface was abandoned

fahime-ghasemi, 2018-10-29:
The issue still exists and app crash with
java.lang.IllegalStateException: failed to get surface
        at android.media.MediaRecorder.getSurface(Native Method)

tibbi, 2018-10-29:
are you using app version 5.0.0?

fahime-ghasemi, 2018-10-29:
yes, I cloned the master branch a few hours ago and app version in build.gradle is 5.0.0

fahime-ghasemi, 2018-10-29:
This is the logcat 

 Process: com.simplemobiletools.camera.debug, PID: 25279
    java.lang.IllegalStateException: failed to get surface
        at android.media.MediaRecorder.getSurface(Native Method)
        at com.simplemobiletools.camera.views.CameraPreview.startRecording(CameraPreview.kt:819)
        at com.simplemobiletools.camera.views.CameraPreview.toggleRecording(CameraPreview.kt:951)
        at com.simplemobiletools.camera.activities.MainActivity.handleShutter(MainActivity.kt:271)
        at com.simplemobiletools.camera.activities.MainActivity.shutterPressed(MainActivity.kt:262)
        at com.simplemobiletools.camera.activities.MainActivity.access$shutterPressed(MainActivity.kt:30)
        at com.simplemobiletools.camera.activities.MainActivity$initButtons$4.onClick(MainActivity.kt:221)
        at android.view.View.performClick(View.java:6261)
        at android.view.View$PerformClick.run(View.java:23752)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6776)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1518)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)

tibbi, 2018-10-29:
hm ok, sadly I cannot reproduce the issue on any of my devices, will try it again someday

neox95, 2019-06-22:
I have exactly the same issue with SimpleCamera 5.0.1 on Android 8.0

closed 2022-10-15 by tibbi