SimpleMobileTools/Simple-Camera #171: Pictures are saved 90° clockwise

mrvanes, 2018-06-17:
All pictures are rotated 90° right relative to the orientation of the phone.
Phone: Alcatel Idol 2 Mini S (6036Y)
Version: Android 6.0.1
Kernel: 3.10.49

tibbi, 2018-06-17:
what app version?

mrvanes, 2018-06-17:
4.1.1 12 june 2018.

tibbi, 2018-06-18:
do other Camera apps work fine? Did it work well in the previous app version, or are you a new user?

Kingwad, 2018-06-20:
Same bug. Moto Z. Oreo 8.0.0.

4.0.0 worked correctly.
4.1.1 images would not register/save
4.1.2 images do save, but rotated 90 degrees

tibbi, 2018-07-06:
@Kingwad @mrvanes  if you are willing to help with debugging it please shoot a mail at hello@simplemobiletools.com. I could send you a test version which would show a message. You just tell me what is written on it, it should help. Thanks

commit 9548d662 (2018-07-09): fix #171, try avoiding capturing rotated photos
closed 2018-07-09 by tibbi