SimpleMobileTools/Simple-Camera #172: All photos are taken with full zoom

Embridioum, 2018-06-18:
Fresh install, latest version from F-Droid. All photos are taken with full zoom. Have tried a bunch of different resolutions, but they all work the same. Sony Xperia XZ1 Compact with Open Camera installed and stock camera disabled.

pelmato, 2018-06-18:
Same here on Xperia X Compact.

tibbi, 2018-06-18:
So during the capture you still the proper preview, just the captured images are fully zoomed in? How about if you try zooming in somewhere around 50% with gestures, is the image still fully zoomed in to 100%?

pelmato, 2018-06-18:
Changing the zoom makes the captured image correct. So this is a temporary workaround: just zoom in and then zoom out. After doing so all photos will be correct (until relaunching the application).

tibbi, 2018-06-18:
alright, Ill probably be able to fix it properly then. Thanks

closed 2018-06-18 by tibbi

pelmato, 2018-06-21:
Ok, I confirm it works now. Thank you !

commit 7afa9695 (2018-06-18): fix #172, set SCALER_CROP_REGION at image capturing only if the user zoomed