SimpleMobileTools/Simple-Camera #176: Add manual shutter speed selection

jlg97, 2018-06-22:
Add the ability to manually select the shutter speed.

Xeoncross, 2019-01-22:
I'd love to have this too. Looks like different phones [handle this differently](https://stackoverflow.com/a/37567225/99923) so you have to parse the camera parameters right?

            Camera mCamera = Camera.open();
            Camera.Parameters params = mCamera.getParameters();

            String flatten = params.flatten();
            String[] paramsSeparated = flatten.split(";");
            for(String cur : paramsSeparated) {
                if(cur.contains("iso") && cur.contains("values")) {
                    ...
                }
            }

tibbi, 2019-01-22:
The Camera app is a nightmate from that point of view. There are so many things done differently depending on the OS version and the manufacturer that there are way too many issues that I cannot reproduce on any of my 5 test devices. Im not really working on the app nowadays, not sure when will I return. Now that some new devices are getting even 4 cameras with different capabilities Im always further from making it a compelling app.

tibbi, 2019-01-22:
Those are some quite seriously outdated solutions. Anyway I prefer handling such things myself to understand it better since it is the core of the app, not just one of the side features.

Xeoncross, 2019-01-22:
I was just looking at them as a way to modify your application to support camera `get/setParameters()`. If you know of a better way, please let me know.