SimpleMobileTools/Simple-Camera #186: All pictures are saved in landscape orientation

tradej, 2018-07-20:
Landscape pictures are saved correctly, portrait pictures are saved rotated 90° clockwise.

App version: 4.1.3 (F-Droid)
Phone: Nokia 5, Android 8.1.0

GarciaPL, 2018-07-28:
There is an option in the settings which is called Flip front camera photos horizontally, maybe you check if it might fix your issue ? Check it out screenshot here https://github.com/SimpleMobileTools/Simple-Camera/issues/187

tradej, 2018-08-03:
I tried that and it doesn't work (I am talking about the rear—main—camera), and even if it did, it would be a bug because the problem is not flipping, it is rotating.

tibbi, 2019-09-03:
should be fixed in 5.1.0, we ill see

AlanSanchezP, 2020-05-17:
I'm also experiencing this bug

App version: 5.2.0.72 from f-droid
Phone: Redmi 6, Android 9, MIUI 11

AlanSanchezP, 2020-05-19:
Just noticed that this is the same bug as #234, my photos orientation follow the same pattern

tibbi, 2022-10-15:
camera has been rewritten from scratch in 5.4.0, so Im closing this

ToxicMushroom, 2023-08-14:
this still persists in 5.7.0 for me

android 13
lineage os 20
viewer used is com.google.gallery3d

landscape rear camera is correct
portrait rear camera is saved rotated 90 deg