SimpleMobileTools/Simple-Camera #187: Save images and videos in separate directories

GarciaPL, 2018-07-28:
Can we add additional option to save videos in separate directories ? I have attached screenshot of setting from Simple Camera which might be extended to allow us to do so.

[image]

GarciaPL, 2018-07-30:
@tibbi Please tell me do you usually accept PR's from other people or do you maintain this project on your own ? Thank you for letting me know in advance.

tibbi, 2018-07-30:
I do accept them if they are good enough and implement a feature that I want to have

GarciaPL, 2018-07-30:
Sounds good. How do you mark tickets which are good candidates to eventually progress with? Just **feature request** ?

tibbi, 2018-07-30:
ye pretty much