SimpleMobileTools/Simple-Camera #190: Inappropriate picture resolutions

MartinGSch, 2018-08-20:
1. On my Sony Xperia Z5 Compact Simple Camera offers different picture resolutions as in the screenshot below - up to 8.3 megapixel.
However the camera is able to produce up to 23 megapixel, please see the 2nd screenshot from the Sony camera app.
1. Similarly the resolutions offered for video are inappropriate - Simple Camera also offers up to 8.3 megapixel here, but for videos this does not make sense. 
1. For both videos and photos a number of low resolutions are offered which I believe are not adequate anymore. They should be removed or at least the number of options reduced, the app does not need five different choices from 0.0 to 0.3 megapixel.

[Picture resolutions offered by Simple Camera app](https://user-images.githubusercontent.com/42523039/44333561-65bcd780-a46f-11e8-9f46-7f9361b6f394.png)

[Picture resolutions offered by Sony camera app](https://user-images.githubusercontent.com/42523039/44333662-c6e4ab00-a46f-11e8-8adf-d3cd0949e9e6.png)

[image]

[image]

Laicure, 2020-04-22:
+1 for 19MP Xperia XZ2
This is only the reason why I don't use the app :(

pokegamer5547, 2021-04-23:
Same issue. I switched to this app to take photos without exif. But now ig I can't really use it because it doesn't take good quality photos as stock because of resolution issue.

closed 2022-10-15 by tibbi