SimpleMobileTools/Simple-Camera #201: Program type already present: android.support.v4.os.ResultReceiver

fahime-ghasemi, 2018-10-29:
I clone the project but can't install it on a device. I got the following error
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':app:transformDexArchiveWithExternalLibsDexMergerForDebug'.
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:103)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:73)
	at org.gradle.api.internal.tasks.execution.OutputDirectoryCreatingTaskExecuter.execute(OutputDirectoryCreatingTaskExecuter.java:51)
	at org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter.execute(SkipUpToDateTaskExecuter.java:59)
	at org.gradle.api.internal.tasks.execution.ResolveTaskOutputCachingStateExecuter.execute(ResolveTaskOutputCachingStateExecuter.java:54)
	at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.execute(ValidatingTaskExecuter.java:59)
	at org.gradle.api.internal.tasks.execution.SkipEmptySourceFilesTaskExecuter.execute(SkipEmptySourceFilesTaskExecuter.java:101)
	at org.gradle.api.internal.tasks.execution.FinalizeInputFilePropertiesTaskExecuter.execute(FinalizeInputFilePropertiesTaskExecuter.java:44)
	at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:91)
	at org.gradle.api.internal.tasks.execution.ResolveTaskArtifactStateTaskExecuter.execute(ResolveTaskArtifactStateTaskExecuter.java:62)
	at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:59)
	at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:54)
	at org.gradle.api.internal.tasks.execution.ExecuteAtMostOnceTaskExecuter.execute(ExecuteAtMostOnceTaskExecuter.java:43)
	at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:34)
	at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker$1.run(DefaultTaskGraphExecuter.java:256)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:336)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:328)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:199)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:110)
	at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker.execute(DefaultTaskGraphExecuter.java:249)
	at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker.execute(DefaultTaskGraphExecuter.java:238)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.processTask(DefaultTaskPlanExecutor.java:123)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.access$200(DefaultTaskPlanExecutor.java:79)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker$1.execute(DefaultTaskPlanExecutor.java:104)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker$1.execute(DefaultTaskPlanExecutor.java:98)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionPlan.execute(DefaultTaskExecutionPlan.java:663)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionPlan.executeWithTask(DefaultTaskExecutionPlan.java:597)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.run(DefaultTaskPlanExecutor.java:98)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:63)
	at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:46)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at org.gradle.internal.concurrent.ThreadFactoryImpl$ManagedThreadRunnable.run(ThreadFactoryImpl.java:55)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.RuntimeException: com.android.builder.dexing.DexArchiveMergerException: Error while merging dex archives: C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\0.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\1.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\2.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\3.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\4.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\5.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\6.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\7.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\8.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\9.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\10.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\11.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\12.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\13.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\14.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\15.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\16.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\17.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\18.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\19.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\20.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\21.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\22.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\23.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\24.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\25.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\26.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\27.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\28.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\29.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\30.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\transforms\dexBuilder\debug\31.jar, C:\Users\IT-10\Documents\Mediana\instagram-requirements\Simple-Camera\app\build\intermediates\tran [... cut ...]
	at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:328)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:199)
	at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:110)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeAction(ExecuteActionsTaskExecuter.java:113)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:95)

tibbi, 2018-10-29:
try pressing Refactor -> Migrate to AndroidX

closed 2018-10-29 by tibbi

Andorer8, 2019-01-09:
I had the same problem.

And in gradle.properties add the below lines (If not created already, create the file in root folder).
android.useAndroidX=true
android.enableJetifier=true