SimpleMobileTools/Simple-Camera #220: kotlin.KotlinNullPointerException

oneofmanyacdcfans, 2019-04-07:
When using messaging apps (tested with ChompSMS and FB Messenger Lite), if one taps the camera button to snap a picture to send, the error in the subject line is given IF the camera app has previously been left in video mode. Camera app comes up with the video record red circle button inste

tibbi, 2019-04-07:
Sadly the Camera app has many issues that I cannot reproduce as too many things differ per manufacturer, OS and OS version.  What Camera app version do you use?

oneofmanyacdcfans, 2019-04-07:
Hi,

I'm using version 4.20 of the app. My device is the 32 bit American variant
of the HTC Desire 510 running the KitKat 4.4.2 based "NonSense 1.03"
somewhat - deodexed ROM. (it's basically the stock ROM debloated and rooted
with some minor visual tweaks). I have your app installed as a system app
and deleted the factory one because yours is faster and produces much
better pictures.

tibbi, 2019-04-07:
ah, that is an ancient device. Not even version 5 is available for you. That wont be receiving any more updates, so sorry, but I cannot fix it. Thanks

closed 2019-04-07 by tibbi