SimpleMobileTools/Simple-Camera #228: Landscape mode not fuctioning properly

puppykickr, 2019-07-01:
When I use portrait mode, it seems fine, but when I go into landscape, the viewfinder view is still vertical, although the camera view is now horizontal.

Here is a link to a video that I saved to my Drive account, showing the problem.

https://drive.google.com/folderview?id=1tVVTNgap6Ayij473Z2p-p31NjdU4uxTb

kllkko, 2020-02-12:
Hi bro,I found the same question,have you solve this problem? could you give me some suggestions ,thanks

puppykickr, 2020-02-13:
@kllkko 
Not yet.
There are many apps in the Simple Mobile Tools collection, and some have been keeping the dev busy.

kllkko, 2020-02-18:
Thanks a lot,I have solved it,just modify one line code in setupCameraOutput() in CameraPreview.kt , then it will be working 

mActivity.runOnUiThread {
                    //mTextureView.setAspectRatio(mPreviewSize!!.height, mPreviewSize!!.width)
                    mTextureView.setAspectRatio(mPreviewSize!!.width, mPreviewSize!!.height)
  }

tibbi, 2020-02-18:
umm, no.

closed 2022-10-15 by tibbi