SimpleMobileTools/Simple-Camera #232: Wide angle lens

uspoma, 2019-09-13:
Hello,

Thanks for your app.

I use an S10 which has also a wide angle lens. Ivan Meler modified your app to use that wide angle lens. Would it be possible to integrate that into your app, say a switch for switching lenses? Preferably also that the app remembers which lens was used last.

Maybe you want to have a look what he has done: [https://github.com/ivanmeler/Simple-Camera](https://github.com/ivanmeler/Simple-Camera)

Thanks for considering that feature request.

Greetings,
Laura

tibbi, 2019-09-13:
hm, he broke quite a few things with that 1 commit. So he implemented some new thing with switching between cameras? I dont really see what real change has been added.

uspoma, 2019-09-14:
I see two things he did. a) He removed the ability to switch to the front camera. b) He set the app to use the wide angle lens only.

So my question would be: Could you make your app supporting the wide angle lens like he did? But with a switch to be able to choose?

tibbi, 2019-09-14:
I definitely dont want to exchange the front camera for wide lens... It might be improved someday in the further future, we will see how.

closed 2019-09-14 by tibbi