SimpleMobileTools/Simple-Camera #235: HDR (option)

violoncelloCH, 2019-10-13:
Hi
I really love this easy to use camera app and would love it even more, if it would allow for an (optional?) HDR mode to take nicer pictures in high contrast scenarios.
What do you think about this? In my opinion it's the only point where it's lacking behind e.g. the Google Camera App...

IzumiSena, 2020-03-30:
Google released cameraX API for third party apps i think you guys can implement this in simple camera here is link (g.co/cameraX) feature like Portrait, HDR, Night, and Beauty available now,,,,

tibbi, 2020-03-30:
Im quite skeptic about CameraX, we will see later if it really helps in any way once it is out of beta and tested by some devs.