SimpleMobileTools/Simple-Camera #252: Cannot save video recording always say "Video recording failed, try using different resolution".

Emulatorer, 2020-01-01:
[image]

###Logcat
[simple_camera_log.txt](https://github.com/SimpleMobileTools/Simple-Camera/files/4013722/simple_camera_log.txt)
###Device Specs
[DevCheck-Specifications.txt](https://github.com/SimpleMobileTools/Simple-Camera/files/4013723/DevCheck-Specifications.txt)

Emulatorer, 2020-01-01:
I tried all the resolution but still failed to save video please fixed this issue 🙏

#happynewyear

Emulatorer, 2020-01-15:
@tibbi any update? 😅

tibbi, 2020-01-15:
not really

Emulatorer, 2020-02-16:
@tibbi where I can download the beta version of this app? or there's a automated build for beta apk?

tibbi, 2020-02-16:
what beta version? What do you expect having there?

Emulatorer, 2020-02-16:
Just for testing purposes :)

Emulatorer, 2020-04-15:
Issue still happen

closed 2020-05-27 by Gamemulatorer