SimpleMobileTools/Simple-Camera #270: Photos occassionally not saved at all

BrendonIrwan, 2020-03-26:
This odd issue happens on specific occasions. I've been trying to narrow down my observations and so far it fails to save photos that are pointed (and focused) on a computer screen.

There is no error message whatsoever, except for an indication that the pictures I took doesn't appear on the gallery thumbnail, Gallery (Simple Gallery), and File Manager (Simple File Manager). I've got plenty of space on memory too.

I get over this by spamming the capture button and eventually one of them would work. I tried different resolutions, disabling EXIF, none of which helped.

This issue never happened with its built-in camera app.

Honor 9 Lite (LLD-L21), Android 9, Simple Camera v5.1.1

ReduxGB, 2021-02-17:
Can reproduce, sometimes produces a cannot save photo error. Other times it doesn't save the photo exif metadata (so its basicly ignored when apps search for images until the device registers them at startup)

BrendonIrwan, 2021-04-08:
To me, the latest version has no issues whatsoever. So follow up with ReduxGB for further details.

ReduxGB, 2021-04-08:
> 
> 
> To me, the latest version has no issues whatsoever. So follow up with ReduxGB for further details.

I am no longer experiencing many issues with the latest version - including this one. Feel free to close this.

closed 2021-04-08 by tibbi

LukeOwlclaw, 2021-05-17:
I had the same problem just recently once, too. It turned out that the photos actually were taken but they just did not show up. Only **after re-starting Android I could see them**. I am wondering whether a re-scan of the SD card as done by https://github.com/jerickson314/sdscanner would have helped, too... But in my case right after having this problem, the build-in photo app stopped taken photos (the "click" button just didn't do anything) so I had to reboot anyways.

BrendonIrwan, 2021-05-18:
I think this one really needs some investigation to know when the problem actually happen. Recently the problem resurfaced again when I was trying to take a picture of my computer screen.