SimpleMobileTools/Simple-Camera #272: No preview ?

Hydro8, 2020-04-06:
Hi

Does simple-camera comes with preview of last capture ?

I only see one picture at the bottom of the settings icon but it's not the preview of the last capture, it's an image of simple-gallery from another folder.

Thank

Hydro8, 2020-04-06:
> it should show the last captured image there under the app settings, maybe
> it misbehaves under some circumstances.

It's never the right image :/

bellegarde-c, 2020-04-29:
Same here, it seems photo are not registered to Android media database.

In Signal, I can share photos from GCam but not from Simple Camera.

anddam, 2020-04-29:
> Same here, it seems photo are not registered to Android media database.

I am experiencing this as well, not sure if it belongs in this issue or not.

I noticed that pictures taken with Camera are not present in Google Photos until I open them, e.g. with Files. Then they are displayed in Photos and known to the media database.

anddam, 2020-05-07:
Also see #268 that possibly has the same cause.

shanemd, 2020-09-03:
I am also experiencing this issue. Here are some details from my phone:

1. In camera mode, no preview is displayed. However, pressing on the place where the preview should be brings up an image viewer, but no image is displayed.
2. In video mode, a preview is displayed, but it's from a random directory not related to the simple camera app. Pressing on it correctly brings up a video player with the video shown in the preview.
3. In both cases, the preview (or invisible preview 'button'), are only displayed after pressing the settings button, opening the settings 'drawer'.

Happy to share more info if it will help solve this issue.

closed 2022-10-15 by tibbi