SimpleMobileTools/Simple-Camera #281: Camera crashes/freezes when being opened from lockscreen

Wurzelmann, 2020-07-21:
When opened from the lockscreen (in my case by a button shortcut), the camera works on the first time, but every time afterwards, it just freezes and shows the interface with a black screen. After manually killing the app, it works again once and then the freezing starts again.

Opening the camera from the launcher works every time, no problem there.

Device: OnePlus 3T
OS: LineageOS 17.1-20200720-nightly-oneplus3

gitwah, 2021-05-23:
Can confirm this behavior. On my device, launching the camera from the lockscreen results in a completely black camera preview with the (unresponsive) UI on top of it. Unlike OP, manually killing the app doesn't help, it never seems to work.

Device: One Plus 7 Pro
OS: LineageOS 18.1, Android 11

hamishmb, 2022-03-27:
Confirmed on Moto G8 Power with Stock Android 11.

Wurzelmann, 2022-04-27:
New phone, same problem:

Device: OnePlus 5T
OS: 18.1-20220426-NIGHTLY-dumpling

tibbi, 2022-10-15:
camera has been rewritten from scratch in 5.4.0, so Im closing this

hamishmb, 2022-10-15:
Please wait until I've confirmed the fix.

Wurzelmann, 2022-10-16:
Same problem persists, sorry to say. :-/

hamishmb, 2022-10-16:
Yeah, let's not get hasty and close all the open bugs. I'm going to go through and try and reproduce some of them soon.

Don't get me wrong, I really like the new UI and it seems much better :)

hamishmb, 2022-10-16:
Interestingly, it's working for me. I'm still on the same device with Android 11. What are you running, Wurzelmann?

Wurzelmann, 2022-10-16:
Same device (OnePlus 5T), but Android 12 now (19-20221011-NIGHTLY-dumpling).

tibbi, 2022-10-16:
havent you uninstalled the stock camera? The current behaviour doesnt seem to be related to Simple Camera at all

Wurzelmann, 2022-10-18:
I had not, but disabled it now and you're right, Simple Camera is working when Stock Camera is disabled. Strange behaviour, but at least we know the culprit now. Thank you for your support, closing now.

closed 2022-10-18 by Wurzelmann

hamishmb, 2022-10-18:
Ah, that explains why it works for me too then.

Should we add a note somewhere for this?

tibbi, 2022-10-18:
no, that isnt actually related to our app at all

hamishmb, 2022-10-19:
It is though, because people will assume your app is broken if they aren't told they need to disable the built-in camera.

hamishmb, 2022-10-23:
My bad, it says in the readme. Is this information presented to users on first launch?

tibbi, 2022-10-24:
no

hamishmb, 2022-10-25:
Okay, I'll file a new issue for that then.