SimpleMobileTools/Simple-Camera #282: [Suggestion] pleas add option to use Camera on Lockscreen (like OpenCamera)

DSDV, 2020-07-26:
I love SimpleCamera as my main and simple camerea ;)  
and use OpenCamera when i need to shoot eBay stuff for example.  
Sadly only OpenCamera supports beeing "overlayed" on the lockscreen :/  
pls add that feature

tibbi, 2020-07-26:
what app version are you using? It should already work just fine

DSDV, 2020-07-26:
latest version from F-Droid

DSDV, 2020-07-28:
Also tried playstore version.
open cammera has an option called "Show camera when locked" which will show the camera above any lockscreen.
in linageos you can enable doubleclick powerbutton to open camera which works also when locked. but sadly not with simple camera

auxcodes, 2022-12-21:
I also am loving Simple Camera, however, the camera stays on when my phone is locked.
This happens on my One Plus Size after opening the camera taking a photo and then locking my phone.
I only notice the issue because my phone gets hot in my pocket and drains the battery because of the issue.

palbr, 2023-09-30:
I just installed "Simple Camera" version 5.7.2 from F-Droid on my Fairphone 3 with e/OS/.

If the mobilephone is locked and I press the powerbutton quickly twice, a vibration confirms my action and the screen turns on. But instead of showing the "Simple Camera" app, I see the keypad to enter my unlock code.

If the mobilephone is unlocked, pressing the powerbutton quickly twice opens the "Simple Camera" app as expected.

"Open Camera" can be started this way. Would be nice, if "Simple Camera" gets an "over lockscreen mode", too.