SimpleMobileTools/Simple-Camera #287: Video Not Working

dorrisowens, 2020-09-24:
Hello, I am using version 5.2.1 while photos works okay but when I click on record video option I get this error
[image]
[image]

Please help me out. java.lang.IllegalStateException: Failed to get surface and java.io.IOException: Prepare Failed are two errors I am facing.

Device : Samsung Galaxy A10
Android Version : 10

dorrisowens, 2020-09-25:
Did You Find Any Fix for this dev ?

tibbi, 2020-09-25:
no, sadly that app can have many issues depending on your device, not sure when will I be able to fix them.

dorrisowens, 2020-09-25:
Ok , I will test it on multiple devices. I have Samsung, Xiaomi and oneplus device. If issue is on all three then I will Inform you.

dorrisowens, 2020-09-25:
Okay, I will wait for fix. If you need any help like logcat or multiple device test then just comment here.

dorrisowens, 2020-09-25:
Hello dev, I found the problem. You have set default video resolution to be highest 3264 × 1836 which my device didn't support. On lowering resolution It works.

dorrisowens, 2020-09-25:
I recommend that in new update try to keep default resolution a little lower.

tibbi, 2020-09-25:
well then it shouldnt list it as an available resolution :/ Ill keep an eye on it. Thanks

dorrisowens, 2020-09-25:
I dont know but it didnt work on avaliable highest resolution. BTW thanks for replying :)

closed 2022-10-15 by tibbi