SimpleMobileTools/Simple-Camera #293: [Pixel 4a] View finder is very laggy, shows images twice or more

ericschdt, 2020-12-08:
Source: F-Droid
Version: Version 5.2.3
Phone: Google Pixel 4a

The viewfinder was supposed to show a keyboard, however that what it looks like:
[image]

When I move the camera I find that the image is updated on a very slowly rate, often with those duplicate compositions. Other apps like OpenCamera work fine.

Fellhahn, 2021-05-12:
I'm seeing identical behaviour on the Pixel 4 XL running CalyxOS.

On the Pixel 4 XL, there are two rear cameras, a 12 MP standard and a 16 MP tele. In Open Camera, these are not surfaced as two different cameras that you can select using the multiple cameras button. Rather when you zoom past a certain limit (2.0x in this case) you are automatically switched from the standard to the tele. (A stupid implementation to be sure).

I have noticed that while Simple Camera will behave incorrectly when first opened and using the standard lens, if i zoom in, forcing the (camera/app/OS?) to use the second sensor, everything behaves normally.

Looking at the specs of the 4 XL, the standard camera is listed as "Dual-Pixel PDAF" while the tele simply says "PDAF" .

Possibly this split image view is a product of trying to communicate with a dual-pixel PDAF sensor.

boggle247, 2021-08-16:
Having the same issue as OP with my pixel 4a running calyxos as well. Stock camera app, opencamera, and Google camera all work fine.

I'd love to use this app bc it's the only one with built in support to remove exif data.

JustACasual, 2021-12-25:
Same here on Pixel 4a with stock Android 12.
Unusable unfortunately.

Thanks for your great work with those simple mobile tools btw. Using some of them and also donated ;)

tibbi, 2023-06-23:
app has been rewritten since then

closed 2023-06-23 by tibbi