SimpleMobileTools/Simple-Camera #299: Can't take a picture

SCOTT-HAMILTON, 2021-06-13:
## Metadata
**Version : 5.3.1**
**Android Version : 8.0.0**
**Phone : Honor 7A AUM-L29**

## Description
When I take a picture, I get this error message : 
`Capture picture android.hardware.camera2.CameraAccessException.CAMERA_ERROR (3): submitRequestList:283: Camera 0: Got error Invalid argument (-22) after trying to submit capture request`

## Screenshot

(in french « une erreur est survenue » = "an error occured")
<details>

[image]

</details>

tibbi, 2021-06-13:
try restarting your device

SCOTT-HAMILTON, 2021-06-14:
I tried but it didn't work, I get the same error. I think the problem isn't with this app though because the stock huawei camera app doesn't work either. I did some cleaning and uninstalled some system apps. I will try to reinstall huawei mobile services.

tibbi, 2021-06-14:
well, it wont be our issue then indeed

closed 2021-06-14 by tibbi

SCOTT-HAMILTON, 2021-06-14:
I tried a factory reset to fix the stock camera app. But it didn't work, so I tried Open Camera and it seems to work.