SimpleMobileTools/Simple-Camera #309: [feature request] an option to set the default image size

07416, 2021-12-07:
Only the JPG compression level can currently be set. Many phones (especially those with custom ROMs installed which replace the proprietary camera software, I assume) capture above 4K-grade photos, taking several MBs of storage space a piece while not specifically benefiting from the high resolution. I commonly scale my camera pictures to about 25% of the original dimensions and currently this has to be done one-by-one in the app.

The AOSP camera app has an option for 'picture size' (in megapixels).

tibbi, 2022-10-15:
camera has been rewritten from scratch in 5.4.0, so Im closing this

closed 2022-10-15 by tibbi

multimob, 2022-12-31:
Could we please reopen this issue?

I run version 5.6.0 and since the rewrite from scratch, all pictures are always captured with the highest resolution. Same for videos, which can easily reach several gigabytes.

Having an option to select the output resolution would be useful here.

tibbi, 2023-01-01:
adding back the resolution picker isnt on the roadmap