SimpleMobileTools/Simple-Camera #310: App is not recognized as Camera app and cannot be used/set as default on Android 12

rugk, 2022-02-28:
## STR

1. Install the app.
2. You cannot actually set the app as the default app for the camera or even choose it when such an intend is sent/other app requests pictures.
3. It e.g. looks like this (you can use "Camera Folder" linked below to reproduce this):
  [image]

Also apps [like Camera Folder](https://f-droid.org/de/packages/de.k3b.android.camerafolder/) won't work like this, because they don't recognize this app as a camera app.

# System

Device: Pixel 4
OS: Android 12, CalyxOS 3.2.0
source: F-Droid
version: v5.3.1

tibbi, 2022-02-28:
This app has recently been discontinued, because it cannot be made reliable enough. You should try some other one.

closed 2022-02-28 by tibbi

rugk, 2022-02-28:
Oh sad to hear.

>  You should try some other one.

Do you have any alternative recommendation?
[On F-Droid I can only find](https://search.f-droid.org/?q=camera&lang=de) more complex (not as simple ones or just not really a normal camera app) 
TBH for my use case it has always worked well enough (I explicitly needed an alternative simple Camera app that saves only small pictures without EXIF metadata without fancy stuff, so it's kinda sad to hear this is discontinued.).

Also I am going to create an issue for "offboarding" steps of this app.

tibbi, 2022-02-28:
I dont have any recommendation, I just use the preinstalled one. Those usually work the most reliably.