SimpleMobileTools/Simple-Camera #311: Mark app as unmaintained in F-Droid + repository

rugk, 2022-02-28:
First of all, you should mark teh app as unmaintained on the app stores. I am not that familiar with F-Droid internals [but I found this doc](https://f-droid.org/en/docs/How_to_Help/#add-information-about-no-longer-maintained-apps), which states you should just add a sentence at the beginning.
I thought you could also trigger that the app could be archived (moved to the archive repo), but that may happen automatically after some time, anyway/maybe cannot or should not be triggered before.

You can also [archive this repository](https://docs.github.com/github/creating-cloning-and-archiving-repositories/archiving-a-github-repository/archiving-repositories), to mark it as unmaintained. People then cannot submit PRs and there is a big warning that this is archived.
Remember to add a note to the Readme before to explain that it is unmaintained, and – if you think this is useful – why and probably mention alternatives.

tibbi, 2022-02-28:
Camera and Clock already has the abandonment note at the top of their descriptions

closed 2022-02-28 by tibbi

rugk, 2022-02-28:
Ah cool okay. So what about archiving this repository?

tibbi, 2022-02-28:
you can still download that app, it just wont be updated

rugk, 2022-02-28:
Yes, you can download it, but what does this have to do with the git repository? As said, you could archive this here on GitHub, so you won't get annoyed by issues etc.

tibbi, 2022-02-28:
nah, I can keep it open in case someone wants to fix something :)