SimpleMobileTools/Simple-Camera #365: F-Droid build failed

linsui, 2022-10-31:
```
Execution failed for task ':app:mergeFdroidReleaseNativeLibs'.
> Could not resolve all files for configuration ':app:fdroidReleaseRuntimeClasspath'.
   > Could not resolve com.github.SimpleMobileTools:Simple-Commons:65cfdf3725.
     Required by:
         project :app
      > Could not resolve com.github.SimpleMobileTools:Simple-Commons:65cfdf3725.
         > Could not get resource 'https://jcenter.bintray.com/com/github/SimpleMobileTools/Simple-Commons/65cfdf3725/Simple-Commons-65cfdf3725.pom'.
            > Could not GET 'https://jcenter.bintray.com/com/github/SimpleMobileTools/Simple-Commons/65cfdf3725/Simple-Commons-65cfdf3725.pom'. Received status code 504 from server: Gateway Time-out
```

This should be resolved to jitpack but jcenter is searched first and break the build. Since JCenter will be shutdown finally, could you please remove it? Thanks!

tibbi, 2022-10-31:
we are not using jcenter

closed 2022-10-31 by tibbi

linsui, 2022-10-31:
Then could you please just remove it from https://github.com/SimpleMobileTools/Simple-Camera/blob/8cdbdad22a9560f788144d70ff4583382d20973f/build.gradle#L8? Thanks!

tibbi, 2022-10-31:
will do someday, just ignore it

linsui, 2022-10-31:
The jcenter is above the jitpack repo. If jcenter has some network issue the dependency resolving will fail as shown above. Anyway, I'll patch it out if it keeps failing.