SimpleMobileTools/Simple-Camera #387: Feature request: Add support for AV1 and AVIF formats

starsforeveryone, 2022-12-10:
Dear developers,

first: Thank you very much for all of your SimpleMobileTools! It is amazing what you have developed and even more amazing that you chose to make it open source and share it with the world.

I have a suggestion on how to make SimpleCamera better. With hardware accelerated AV1 decoding support already implemented in many cpus/gpus, and hardware encoding support hitting the consumer market this christmas, web browsers adapting the format and so on - it seems logical that this file format will become widespread and probably even the next industry standard. Qualcomm will add hardware support for it in their 2023 mobile phone chips, too.

Many open source projects such as Krita, GIMP, ... are already implementing support for it.

Would it be possible to add support for this codec to SimpleCamera, so that the user can take videos in AV1 codec and pictures in AVIF codec? These codecs save a lot of space, which seems to be a good idea especially on mobile phones.

Thank you very much! :)

tibbi, 2022-12-10:
guess I can keep it here, but dont expect having it anytime soon

starsforeveryone, 2022-12-11:
Thank you!

I know it's difficult. JPEG XL is not supported by Android yet, and maybe never will. So you as a developer cannot implement it. AV1 video recording doesn't make sense in a camera app unless the device has AV1 hardware encoding support, which will only be available in flagship smartphones from 2023 on. I can see that this is too much work or even impossible for you as a developer. I mainly wanted to raise awareness that these changes are coming to the industry.

AVIF picture recording is a lot easier. Android 12 and later already have built-in support for it. I just checked, SimpleGallery supports AVIF files on my Android 12 phone! 😎 The structures in Android seem to already exist and work. Implementing it should therefore be way, way easier.

Therefore, it would be nice if you could add AVIF as a picture format to SimpleCamera. You would also be the first app to support it.

Thank you very much, regardless of whether you will find time for it.