SimpleMobileTools/Simple-Camera #399: Private by default camera (EXIF)

staticdev, 2023-01-18:
Hi @tibbi, I am really glad we have these amazing Simple Apps and also happy we can [enable/disable EXIF](https://github.com/SimpleMobileTools/Simple-Camera/issues/39). One suggestion I have is to make it *disabled by default* to protect user privacy. I think it should be opt-in since today many people share photos without thinking if they are sharing location or phone data in it. This already happens in other camera apps such as Libre Camera. If you agree I am happy to make my first PR to the project.

tibbi, 2023-01-18:
not really, I dont plan adding that. There is no location in the photos. Thanks though

closed 2023-01-18 by tibbi

staticdev, 2023-01-18:
@tibbi not 100% sure if you understood what I suggested. There is nothing to add, instead just come with the toggle off by default. Even if there is no location it is still phone metadata that should not be shared if the used to not want or thing about that.

tibbi, 2023-01-18:
I dont want to disable it by default

barrelful, 2023-01-20:
Always good to see the attitude of the main developer of an app towards privacy of it's users. Just lost my install and star in you repo with this one. I will also start recommending other apps to my colleagues.