SimpleMobileTools/Simple-Camera #412: Photos taken in Landscape appear as Portrait in all Gallery Apps

Starcr0ss, 2023-05-13:
Hi,
When I take a photo in landscape in Simple Camera v5.6.2, the photo appears in gallery apps as portrait. This does not happen in the default Motorola camera app. I have a Moto G Stylus 2021.
Here is an example photo.
[image]

tibbi, 2023-05-14:
how does it work in other downloaded galleries?

Starcr0ss, 2023-05-14:
> how does it work in other downloaded galleries?

Here it is in Simple Gallery, Google Photos, and Camera Roll X
[image]
[image]
[image]

tibbi, 2023-05-14:
so the issue apparently isnt with our app, it is in the way file rotation is stored

closed 2023-05-14 by tibbi

Starcr0ss, 2023-05-17:
Very interesting, the problem seems to no longer occur in V5.6.3!