SimpleMobileTools/Simple-Camera #457: [5.7.2] When switching to Video: "An error occured while configuring th…"

Shohreh, 2024-01-30:
Hello,

I encoutered a more serious issue with release 5.7.2 (on the same Xiaomi Redmi Note 5 running Android 13: When switching from Photo to Video, it displays the following error message: "[An error occured while configuring th…"

Switching back to photo doesn't hide the message: Only killing the app does.

So at this point, I can't use the app to record a video.

Cheers,

[image]

inson1, 2024-01-30:
@Shohreh SMT was sold to ad company. Fork is here https://github.com/FossifyOrg

Shohreh, 2024-01-30:
Good to know but… how can I install it? With its plain name, it's hard to find in F-Droid.

inson1, 2024-01-30:
@Shohreh I wouldnt call it plain name. Its unique imo. Camera isnt yet released, but few app already are https://search.f-droid.org/?q=fossify&lang=en Unfortunately the Camera isnt the most favorite out of the group.

Shohreh, 2024-01-31:
Finding a simple, free-beer, bug-free, GPS-capable camera application is proving to be much harder than expected :-/

Google: Pics not geotagged

Simple Camera, https://github.com/FossifyOrg/Camera : Not yet released

Simple Camera 5.7.2: Video doesn't work, besides "right top" bug

Libre Camera 1.9.1 : Picts not geo-tagged
​​
Open Camera: Too many features (don't need three cameras IDs), weird colors, slow to start and take picture

inson1, 2024-01-31:
@Shohreh use the one from F-droid, till Fossify version is out

Shohreh, 2024-02-01:
What app are you talking about?

The only app I see when searching through the F-Droid client is Simple Camera 5.7.2… whose video mode doesn't work.

[image]

inson1, 2024-02-02:
@Shohreh I thought that you are using the app at least for photos, if videos dont work
PS for me it works, so its probably device specific (I also have Xiaomi)

Shohreh, 2024-02-02:
Thanks. I'll pass and try to find an app that works and is available now.